trabucayre/openFPGALoader #511: Cannot program Tang Nano 20k: CRC Error / ID Verify Failed

Peter-van-Tol, 2025-01-27:
I'm trying to program a blink example on my Tang Nano 20k, however the sketch won't load. During the process of flashing to SRAM, I get the following errors `CRC Error` and `ID Verify Failed` when using the following command:
  `openFPGALoader -b tangnano20k  Blink.fs --verbose-level 2 --freq 2500000`

The full output of the command is
```
empty
Jtag frequency : requested 2.50MHz   -> real 2.50MHz  
Raw IDCODE:
- 0 -> 0x0000081b
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
        idcode 0x81b
        manufacturer Gowin
        family GW2A
        model  GW2A(R)-18(C)
        irlength 8
File type : fs
Parse file Parse Blink.fs: 
checksum 0x94cb
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 1342
LoadingRate: 2500000
ProgramDoneBypass: OFF
SPIAddr: 00000000
SecurityBit: ON
idcode: 0000081b
before program sram: displayReadReg 002005fa
        Bad Command
        Timeout
        Reserved4
        Memory Erase
        Preamble
        System Edit Mode
        Program SPI FLASH directly
        Non-JTAG configuration is active
Erase SRAM before erase sram: displayReadReg 02fa0000
        FLASH lock
pollFlag: 80000000 (0)
pollFlag: 1005fa05 (0)
pollFlag: 7d000000 (0)
pollFlag: 20 (0)
pollFlag: 2005fa (80)
pollFlag: 2fa0000 (0)
pollFlag: 80000000 (0)
pollFlag: 2005 (0)
pollFlag: 7d05fa00 (0)
pollFlag: 2005fa (20)
after erase sram: displayReadReg 02fa0000
        FLASH lock
pollFlag: 80000000 (0)
Load SRAM before write sram: displayReadReg 00002005
        CRC Error
        ID Verify Failed
        Done Final
Load SRAM: [==================================================] 100.00%
Done
after write sram: displayReadReg 1005fa00
        Reserved9
        Bypass
        Gowin VLD
        Done Final
        Security Final
        Ready
        POR
        FLASH2 lock
DONE
after program sram: displayReadReg 7d000000
```

I'm using `openFPGALoader` version 0.13.1 on Debian. I've tried:
- multiple bitstreams, both generated with the open-source toolchain `apicula` and the Gowin Eda;
- flashing with Gowin Programmer on Windows, this works, so the both bitstreams are verified to be working;

Can somebody give me some guidance to solve this error and/or to flash the Tang Nano 20K, so I can move the whole process to the open source toolchain on Linux?

trabucayre, 2025-01-27:
Could you share the .fs and if possible the project to reproduces, locally, this issue?
Bad CRC mean the bitstream was not correctly loaded. ID verify Failed is when the fs is build from a device different than the target.

Thanks

Peter-van-Tol, 2025-01-27:
Please find the file and project attached: 

[Blink.zip](https://github.com/user-attachments/files/18564471/Blink.zip)

trabucayre, 2025-01-27:
Not tried to rebuild the project but able to load the bitstream without issue.
But the bitstream read at boot time is valid, maybe your flash has a corrupted bitstream and maybe a missing step is required when the FPGA is in bad state.
Could you:
1. provides output for `sudo lsusb -vv -d 0403:6010` command
2. writing a bitstream into the flash with Gowin tools to have the FPGA in a good state

Peter-van-Tol, 2025-01-27:
Output for `sudo lsusb -vv -d 0403:6010 command`

```
Bus 003 Device 002: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6010 FT2232C/D/H Dual UART/FIFO IC
  bcdDevice            7.00
  iManufacturer           1 SIPEED
  iProduct                2 USB Debugger
  iSerial                 3 2024122312
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0037
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 USB Debugger
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 USB Debugger
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

trabucayre, 2025-01-27:
Ok the firmware for the USB<->JTAG is more recent than mine. Maybe something here?
If your board with a valid flash's content continue to fails with openFPGALoader I have to try to update the firmware to see if something is wrong here.

Peter-van-Tol, 2025-01-27:
Unfortunately it fails when I've a working sketch on the board.

```
empty
Jtag frequency : requested 2.50MHz   -> real 2.50MHz  
Raw IDCODE:
- 0 -> 0x0000081b
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
        idcode 0x81b
        manufacturer Gowin
        family GW2A
        model  GW2A(R)-18(C)
        irlength 8
File type : fs
Parse file Parse Blink.fs: 
checksum 0x94cb
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 1342
LoadingRate: 2500000
ProgramDoneBypass: OFF
SPIAddr: 00000000
SecurityBit: ON
idcode: 0000081b
before program sram: displayReadReg 302005fa
        Bad Command
        Timeout
        Reserved4
        Memory Erase
        Preamble
        System Edit Mode
        Program SPI FLASH directly
        Non-JTAG configuration is active
Erase SRAM before erase sram: displayReadReg 02fa0000
        FLASH lock
pollFlag: 80000060 (0)
pollFlag: 1005fa05 (0)
pollFlag: 7d000000 (0)
pollFlag: 6020 (0)
pollFlag: 302005fa (80)
pollFlag: 2fa0000 (0)
pollFlag: 80000060 (20)
after erase sram: displayReadReg 00602005
        CRC Error
        ID Verify Failed
        Done Final
pollFlag: 7d05fa00 (0)
Load SRAM before write sram: displayReadReg 302005fa
        Bad Command
        Timeout
        Reserved4
        Memory Erase
        Preamble
        System Edit Mode
        Program SPI FLASH directly
        Non-JTAG configuration is active
Load SRAM: [==================================================] 100.00%
Done
after write sram: displayReadReg 02fa0000
        FLASH lock
FAIL
```

It still has the `CRC Error` and the `ID Verify Failed` errors. Also it fails the `FLASH lock`.

Is the firmware version of my board `2024122312`? And what is your firmware version?

trabucayre, 2025-01-28:
Ok, so the problem isn't related to the current FPGA's state.

My firmware's version is `2023030621`.
Have to see how to update the BL device and where to find firmware.

hoglet67, 2025-01-30:
Just to say I'm also experiencing the same issue with the new 2024122312 BL616 firmware in a recently purchased board. It's not bitstream specific.

Programming with the Gowin programmer works.

The other weirdness is the serial baud rate is confused. If you have a bitstream that includes a serial UART running at 115200 (e.g. Litex) you have to connect at 460800 (i.e. 4x the speed). So something is definitely messed up with that firmware version.

trabucayre, 2025-01-31:
@hoglet67 thanks! True I have already seen this issue too (but unfortunately not with one off boards I have).
Some contribs have been done to add work around for this firmware's version but it look like not enough.

I have to ask @sipeed where to find the firmware to update BL616 and/or if firmware has been updated fixed since this version.

Peter-van-Tol, 2025-02-13:
Thank you very much!

I've the time to try this out tonight.

trabucayre, 2025-02-13:
Thanks to share these informations!
I will try to downgrade my board to see if it's possible to see how to avoid this failure.
I have a recent board with a firmware 2025xxx and don't see any issues: so it make sense to update boards with a buggy firmware and to add an entry into documentation with link and instructions for people with a bad firmware.

hoglet67, 2025-02-13:
> I will try to downgrade my board to see if it's possible to see how to avoid this failure.

I don't think the buggy 2024122312 firmware has ever been posted online by SiPEED.

trabucayre, 2025-02-14:
@hoglet67 In fact, updating the firmware seems easy and fix issues, so I'm not sure its make sense to spent time to try fixing that -> it's better to use the new one.

@Peter-van-Tol Thanks for your feedback. FAQ must be updated to adds a note for users with the same issue.

Peter-van-Tol, 2025-02-15:
Reopened this issue for updating the FAQ.

PR #515 (2025-02-15): Add entry about updating firmware TangNano20K
Closes #511
closed 2025-02-15 by trabucayre