trabucayre/openFPGALoader #513: Color Light 5a-75b board version 8.2

kenkit, 2025-02-11:
I had to modify the usb vid and pid to match
```
diff --git a/src/anlogicCable.cpp b/src/anlogicCable.cpp
index d6f6ffb..e094ea9 100644
--- a/src/anlogicCable.cpp
+++ b/src/anlogicCable.cpp
@@ -17,7 +17,7 @@
 
 using namespace std;
 
-#define ANLOGICCABLE_VID 0x0547
+#define ANLOGICCABLE_VID 0x336c
 #define ANLOGICCABLE_PID 0x1002
 
 #define ANLOGICCABLE_CONF_EP  0x08
ken@ken-HP-ProBook-640-G1:/tmp/openFP
```

my lsusb 
Bus 003 Device 017: ID 336c:1002 Anlogic USB-JTAG-Cable

```
sudo openFPGALoader -b colorlight  -c anlogicCable  --dump-flash  s
empty
Jtag frequency : requested 6000000Hz -> real 6000000Hz
JTAG init failed with: Unknown device with IDCODE: 0xffff8803 (manufacturer: 0x401 (), part: 0x7f vers: 0xf
```

kenkit, 2025-02-11:
This is the board https://github.com/q3k/chubby75/blob/master/5a-75b/hardware_V8.2.md

trabucayre, 2025-02-12:
I think the board isn't the problem here: ECP5 is know by openFPGALoader and I have already uses this board (or a similar model) with any issue.
It's a bit weird to have to change the Vendor ID for the cable: could you provides more informations about the JTAG cable you use (a link?) I think this issue is related to this device, maybe some changes into the protocol or something like that.

kenkit, 2025-02-13:
I used stm32 bluepill with a closed source [firmware ](https://github.com/AnlogicInfo/anlogic-usbjtag)
originally it was not being detected without modifying the source code.

kenkit, 2025-02-13:
Also my board is v8.2 I think it uses a different fpga compared to older boards

trabucayre, 2025-02-14:
Thanks for the link: I see in commit messages one about new vendor ID. Code must be updated to support the old and the new VID.

I have a bluepill: I will try to reproduce. IDCODE read look bad, maybe some changes in the firmware's code has introduce a regression.

kenkit, 2025-02-15:
I am also trying to get urjtag to work then see what happens.

kenkit, 2025-04-20:
I have re-cloned the repo and rebuilt, the result is the same 

[image]

kenkit, 2025-04-20:
Are you sure you have the same board as mine ? 
I think this board is not similar my board is [v8.2](https://github.com/q3k/chubby75/blob/master/5a-75b/hardware_V8.2.md)

trabucayre, 2025-04-27:
Boards are all based on ECP5, could you recheck way the adapter is connected to the FPGAs: sometimes dupont wires may introduces issues.

kenkit, 2025-05-07:
I will try again with different wires

kenkit, 2025-05-31:
FInally it worked.
I was going to connect to my Oscilliscope then this is what I got.
Had to change the wires since it was through the bread board

[image]

Memory
[image]

closed 2025-05-31 by kenkit

kenkit, 2025-05-31:
This is the bitstream I dumped.
Not sure if I got it all

[image]

kenkit, 2025-06-01:
@trabucayre It seems my boards pinout has changed since the blink example does nothing after programming.
How do you go about finding the correct pinout of such boards ?
Is there some code we can use to bruteforce all pins ?

[image]