trabucayre/openFPGALoader #518: Errors after failed write

zaun, 2025-02-22:
I was writing to the FPGA like I've done many times. This time the USB cable came out mid-write and the write failed. When I plugged it back in and tried to re-load it now fails every time with "flash chip unknown: use basic protection detection"

I know from the last successful load it's a "Detected: Winbond W25Q64 128 sectors size: 64Mb" flash. Is there a way I can fix this or did I brick my board?

Thanks in advance.

Full Log:

```
openFPGALoader --verbose-level 2 -b tangnano20k out.fs -f                                                                                                                                1 ↵

empty
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Raw IDCODE:
- 0 -> 0x0000081b
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
	idcode 0x81b
	manufacturer Gowin
	family GW2A
	model  GW2A(R)-18(C)
	irlength 8
File type : fs
Parse file Parse out.fs: 
checksum 0x4ffe
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 1342
LoadingRate: 2500000
ProgramDoneBypass: OFF
SPIAddr: 00fff000
SecurityBit: ON
idcode: 0000081b
after program flash: displayReadReg 00004421
	CRC Error
	Memory Erase
	Non-JTAG configuration is active
	Security Final
Erase SRAM before erase sram: displayReadReg 00004421
	CRC Error
	Memory Erase
	Non-JTAG configuration is active
	Security Final
pollFlag: 40a1 (80)
pollFlag: 4081 (0)
pollFlag: 4081 (0)
pollFlag: 4081 (0)
pollFlag: a1 (20)
after erase sram: displayReadReg 000000a1
	CRC Error
	Memory Erase
	System Edit Mode
pollFlag: 421 (0)
after erase sram: displayReadReg 00000421
	CRC Error
	Memory Erase
	Non-JTAG configuration is active
DONE
Jtag probe limited to %d MHz6000000
Jtag frequency : requested 10.00MHz  -> real 6.00MHz  
ff ff ff ff read ffffffff
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
ff ff ff ff read ffffffff
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
RDSR : ff
WIP  : 1
WEL  : 1
BP   : f
TB   : 1
SRWD : 1
RDSR : fc
WIP  : 0
WEL  : 0
BP   : f
TB   : 1
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
Error: write to flash failed
Error: Failed to program FPGA: Error: write to flash failed
```

zaun, 2025-02-22:
Sorry, I fixed this by using the gowin IDE to program the board once. Now openFPGALoader is working again.

closed 2025-02-22 by zaun

trabucayre, 2025-02-22:
To avoid futur error maybe you could check that: https://trabucayre.github.io/openFPGALoader/guide/troubleshooting.html#cannot-flash-tang-nano-20k-issue-251

Ok it's for tang-nano but maybe the firmware of your board must be also updated