trabucayre/openFPGALoader #519: MPSSE command sent before MPSSE enabled

harbaum, 2025-02-26:
Hi. In SPI mode my USB monitoring utility reports two times that MPSSE is being used to set ports before MPSSE mode is actually enabled:

```
USB Vendor request 0x0 value:0 index:1 Reset -> Reset SIO
USB Vendor request 0x3 value:1250 index:513 Set baud rate -> divisor=0x20104e2 9600.0
USB Vendor request 0x3 value:49256 index:513 Set baud rate -> divisor=0x201c068 243.62514211466623
USB EP2 BULK OUT[3]: 80 48 cb
Warning: not in mpsse mode 
USB EP2 BULK OUT[3]: 80 48 cb
Warning: not in mpsse mode 
```

I tracked one of these down and it's this line:
https://github.com/trabucayre/openFPGALoader/blob/2ff215dffffeaca0ecb6f8749e68c3cb41309914/src/ftdispi.cpp#L103
being called right before enabling MPSSE in this line:
https://github.com/trabucayre/openFPGALoader/blob/2ff215dffffeaca0ecb6f8749e68c3cb41309914/src/ftdispi.cpp#L109

This is a problem since setting GPIOs is done through MPSSE:
https://github.com/trabucayre/openFPGALoader/blob/2ff215dffffeaca0ecb6f8749e68c3cb41309914/src/ftdipp_mpsse.cpp#L614-L627

And there's a second case like this probably hidden somewhere in 
https://github.com/trabucayre/openFPGALoader/blob/2ff215dffffeaca0ecb6f8749e68c3cb41309914/src/ftdispi.cpp#L102-L109

This isn't really a problem as this will probably only cause some random serial data being sent through the UART. But this may still be unexepected and confusing ...

trabucayre, 2025-02-26:
Good catch! Its now fixed.
Thanks!

harbaum, 2025-02-27:
Indeed fixed! Thanks!