trabucayre/openFPGALoader #524: SPIFlash buffer size handling broken

harbaum, 2025-02-28:
Sorry for constantly bombarding you with issues ... but ...

In ft2232_spi_wr_and_rd() on a write the max_xfer is set to 4096:
https://github.com/trabucayre/openFPGALoader/blob/564485036e222b80bc19afe638a27f688c04903f/src/ftdispi.cpp#L171

This is used to limit the xfer value to something below max_xfer.  Subsequently mpsse_store() is called and finally mpsse_write(). The return values of mpsse_write() is compared with xfer len+3:

https://github.com/trabucayre/openFPGALoader/blob/564485036e222b80bc19afe638a27f688c04903f/src/ftdispi.cpp#L218

The problem is that mpsse_store() will internally already call mpsse_write() whenever the buffer is full e.g. here:
https://github.com/trabucayre/openFPGALoader/blob/564485036e222b80bc19afe638a27f688c04903f/src/ftdipp_mpsse.cpp#L493

Upon return mpsse_write() returns the number of bytes written. The problem is the the previous comparison with xfer_len+3 will fail if mpsse_store() has already emptied the buffer and called mpsse_write itself. The final mpsse_write() will then only return the length of the last chunk.

This is triggered when writing to a full speed device with endpoints of 64 bytes. Data to flash seems to be written in chunks of 256 bytes which does fit into the 4096 bytes max_xfer limit but more importantly it also fits into the buffer of a high speed usb device with 512 bytes endpoint buffers. I am using a full speed device and thus this triggers.

This will also trigger if some upper layer tries to write > 512 bytes in a chunk. This just never seems to be the case. 

May I suggest to just change 
https://github.com/trabucayre/openFPGALoader/blob/564485036e222b80bc19afe638a27f688c04903f/src/ftdispi.cpp#L171
to
`uint32_t max_xfer = _buffer_size;`

Or just use _buffer_size instead of max_xfer 

Unfortunately this doesn't fix all issues and once this is done I see another `*** buffer overflow detected ***: terminated`. I'll investigate that further ...

harbaum, 2025-02-28:
Edit: To trigger this artificially it should be sufficient to temporarily set `_buffer_size` to 64. That should not cause any harm and will only cause smaller chunks to be sent then possible. But this should trigger this problem on high speed devices as well.

harbaum, 2025-02-28:
Oh .. there's more in ft2232_spi_wr_and_rd() 

It allocates a buffer of size max_xfer but copies the MPSSE command to byte 0, the xfer len to bytes 1 and 2 and then adds xfer_len data bytes. But if xfer_len is max_xfer then this overflows the buffer by exactly the three mpsse header bytes ...

trabucayre, 2025-03-02:
Good catch and analyze: must be fixed by https://github.com/trabucayre/openFPGALoader/commit/ecb22b2eeb2af6699df90bcac34b48e862e0aa31

Thanks

harbaum, 2025-03-02:
Yes, thanks. That fixed the problem!

closed 2025-03-02 by harbaum