trabucayre/openFPGALoader #525: olimex_gatemateevb: Failed to program FPGA: Read ID failed

Martoni, 2025-03-05:
Hello,
I have two computers I use to develop on olimex  GateMateA1-EVB.

One computer is running Ubuntu : 

```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.5 LTS"
```

With following kernel : 
```
Linux  6.8.0-110052-tuxedo #53~22.04.1tux1 SMP PREEMPT_DYNAMIC Wed Jan 15 22:55:57 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```

The board is detected on USB : 

```
$ dmesg
[mars 2 09:02] usb 1-2.3: USB disconnect, device number 50
[  +1,964257] usb 1-2.3: new full-speed USB device number 53 using xhci_hcd
[  +0,101816] usb 1-2.3: New USB device found, idVendor=1209, idProduct=c0ca, bcdDevice= 1.10
[  +0,000016] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000007] usb 1-2.3: Product: DirtyJTAG
[  +0,000007] usb 1-2.3: Manufacturer: Jean THOMAS
[  +0,000006] usb 1-2.3: SerialNumber: 2600942311111956
[  +0,052232] cdc_acm 1-2.3:1.1: ttyACM0: USB ACM device
```

I can't flash bitstream from this [project](https://github.com/Martoni/Martoni_Pcb_collection/tree/main/glm5va/rtl/fourleds) : 

```Shell
$ openFPGALoader -V
openFPGALoader v0.13.1
$ make jtag-flash
openFPGALoader  -b olimex_gatemateevb -f --cable dirtyJtag --verify fourleds_00.cfg
empty
write to flash
Board default cable overridden with dirtyJtag
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Error: Failed to program FPGA: Read ID failed
make: *** [../config.mk:45: jtag-flash] Error 1
```

Just configuring fpga (no flashing) is executed with no error, but no leds blinking : 
``` 
$ make jtag
openFPGALoader  -b olimex_gatemateevb --cable dirtyJtag fourleds_00.cfg
empty
Board default cable overridden with dirtyJtag
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Load SRAM via JTAG: [==================================================] 100.00%
Done
```

On the second computer : 
Computer is running linux Mint :

```
$ cat /etc/lsb-release
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=21
DISTRIB_CODENAME=vanessa
DISTRIB_DESCRIPTION="Linux Mint 21 Vanessa"
```

Olimex board is detected on the USB (the same usb cable) :

```
$ dmesg
[683460.535479] usb 1-1.4.3: new full-speed USB device number 115 using 
ehci-pci
[683460.650609] usb 1-1.4.3: New USB device found, idVendor=1209, 
idProduct=c0ca, bcdDevice= 1.10
[683460.650624] usb 1-1.4.3: New USB device strings: Mfr=1, Product=2, 
SerialNumber=3
[683460.650631] usb 1-1.4.3: Product: DirtyJTAG
[683460.650635] usb 1-1.4.3: Manufacturer: Jean THOMAS
[683460.650639] usb 1-1.4.3: SerialNumber: 2600942311111956
[683460.652670] cdc_acm 1-1.4.3:1.1: ttyACM0: USB ACM device
```

openFPGALoader version is the same :

```
$ openFPGALoader -V
openFPGALoader v0.13.1
```

Kernel is :

```
$ uname -a
Linux  5.15.0-126-generic #136-Ubuntu SMP Wed Nov 6 10:38:22 UTC 2024 
x86_64 x86_64 x86_64 GNU/Linux
```

Configure and flash board with the same project works perfectly !

Configure:

```
$  make jtag
openFPGALoader  -b olimex_gatemateevb fourleds_00.cfg
empty
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Load SRAM via JTAG: [==================================================] 
100.00%
Done
```

Flash :
```
$  make jtag-flash
openFPGALoader  -b olimex_gatemateevb -f --verify fourleds_00.cfg
empty
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
JEDEC ID: 0xba6015
Detected: Zetta ZD25WQ16CSIGT 32 sectors size: 16Mb
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00010000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
```

If you have a clue ?

Martoni, 2025-03-05:
I just tested back on the second computer this evening and I now have the same problem for flashing : 

```
openFPGALoader  -b olimex_gatemateevb -f --verify fourleds_00.cfg
empty
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Error: Failed to program FPGA: Read ID failed
make: *** [../config.mk:44 : jtag-flash] Erreur 1
```

The fpga configuration (make jtag) is keep on working.

trabucayre, 2025-03-05:
I have to add a SPI Flash: my board was shipped without.
Have you tried with the gatemate EVB? Just to see if this issue is related to the gatemate driver or to the PICO-DirtyJTAG.
Thanks

trabucayre, 2025-03-06:
Indeed, seems to have an issue with the olimex board:
CologneChip Gatemate EVB:
```bash
openFPGALoader -b gatemate_evb_jtag -f --detect --verbose-level 2
empty
write to flash
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
Raw IDCODE:
- 0 -> 0x20000001
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
        idcode 0x20000001
        manufacturer colognechip
        family GateMate Series
        model  GM1Ax
        irlength 6
File type : 
Read Flash c2 28 17 c2 read c22817c2
JEDEC ID: 0xc22817
Detected: Macronix MX25R6435F 128 sectors size: 64Mb
c2 28 17 c2 read c22817c2
JEDEC ID: 0xc22817
Detected: Macronix MX25R6435F 128 sectors size: 64Mb
RDSR : 0x00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
```

olimex Gatemate EVB:
```bash
openFPGALoader -b olimex_gatemateevb -f --detect --verbose-level 2
empty
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Raw IDCODE:
- 0 -> 0x20000001
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
        idcode 0x20000001
        manufacturer colognechip
        family GateMate Series
        model  GM1Ax
        irlength 6
File type : 
Read Flash 0 0 0 1 Fail
Read ID failed
```

trabucayre, 2025-03-06:
When gatemate isn't configured:
```bash
openFPGALoader -b olimex_gatemateevb -f /tmp/blinky_00.cfg.bit --verbose-level 2
empty
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
Raw IDCODE:
- 0 -> 0x20000001
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
        idcode 0x20000001
        manufacturer colognechip
        family GateMate Series
        model  GM1Ax
        irlength 6
File type : bit
1 60 18 ff read 16018ff
JEDEC ID: 0x016018
Detected: spansion S25FL128L 256 sectors size: 128Mb
RDSR : 0x00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0

Configuration Register
RDCR    : 00
SUS_D   : 0
CMP_NV  : 0
LB      : 0
QUAD_NV : 0
SRP1_D  : 0
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00010000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
```

It seems the problem is somewhere here

trabucayre, 2025-03-07:
I have added two commit:
- https://github.com/trabucayre/openFPGALoader/commit/3c7324d14d6c5fcf982d5cc65c72ebbae7e05c49
- https://github.com/trabucayre/openFPGALoader/commit/3cf558ebe2f878bd598e883d2f7783363efdb172
to drive reset pin with DirtyJTAG cable.
You have to update (if needed) the PICO firmware (https://github.com/OLIMEX/GateMateA1-EVB/tree/main/SOFTWARE/dirtyJtag).

I have soldered an SPI Flash, configured switch correctly but the FPGA seems ignore SPI Flash content... So if you could confirm the fix is fine...

Martoni, 2025-03-07:
-  I updated DirtyJTAG "cable"
-  I pull openFPGALoader master
- I recompiled it (cd build; make ../; sudo make install)
-  And it's work \o/

```shell
 make jtag-flash
openFPGALoader  -b olimex_gatemateevb -f --verify ledsbasic_00.cfg
empty
write to flash
Jtag frequency : requested 6000000Hz -> real 6000000Hz
JEDEC ID: 0xba6015
Detected: Zetta ZD25WQ16CSIGT 32 sectors size: 16Mb
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00010000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
```

Thank you very much :)

closed 2025-03-07 by Martoni

trabucayre, 2025-03-08:
Great! Thanks for your feedback.