trabucayre/openFPGALoader #531: Support for MAX10 10M08SAE144C8G

snevindsouza, 2025-03-24:
Hi, Is this chip supported?

I was testing the max10 FPGA and I tried this and I got this error

```
snevin@WSL:/mnt/c/Users/snevi/Desktop/FPGA/max10text/output_files$ openFPGALoader `-c` usb-blaster --probe-firmware /mnt/c/intelFPGA_lite/20.1/quartus/bin64/blaster_6810.hex -v test1.svf
empty
Read error: -7
JTAG init failed with: Unknown device with IDCODE: 0x2f2e2f2e (manufacturer: 0x797 (), part: 0x79 vers: 0x2
snevin@REDRUM:/mnt/c/Users/snevi/OneDrive/Desktop/Chamber_of_secrets/FPGA/max10text/output_files$
```

I tried with Cyclone 5 - 5CSEBA6U23I7NDK and it works fine

```
snevin@WSL:/mnt/c/Users/snevi/Desktop/FPGA/Cyclone5test/output_files$ openFPGALoader  -c usb-blasterII --probe-firmware /mnt/c/intelFPGA_lite/20.1/quartus/bin64/blaster_6810.hex -v output_file.rbf
empty
USB-Blaster II firmware version: 1.39
found 2 devices
index 0:
        idcode   0x4ba00477
        type     ARM cortex A9
        irlength 4
index 1:
        idcode 0x2d020dd
        manufacturer altera
        family cyclone V Soc
        model  5CSE*A6/5CSX*6
        irlength 10
File type : rbf
Load SRAM: [==================================================] 100.00%
Done
```

Please help me with this 
Thank you

snevindsouza, 2025-03-25:
I don't know what's the exact information you're looking for. However, it's a custom-made FPGA evaluation board but it's very similar to **Intel 10M08 Evaluation Kit**. Full info about the board on the following links:

```
Intel- https://www.intel.com/content/www/us/en/products/details/fpga/development-kits/max/10m08-evaluation-kit.html
Terasic- https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&No=708
```

**Part number: 10M08SAE144C8G
JTAG: Altera USB-Blaster (not USB-BlasterII)
JTAG Speed: 6 MHz**

Also I did try multiple times and the same error has been thrown.
Do I need to modify any source code? If yes please guide me through.

Thank You

jasonrodrig, 2025-03-28:
Hi, I'm facing a similar issue with the same device and USB-blaster. Please let me know how to fix it.

trabucayre, 2025-03-28:
Weird: I have used this board for developing MAX10 support with an USB-Blaster without observing any issues.
Is your USB-Blaster is an official cable or a clone? Sometime clones are not correctly working (I have to get one to see how to fix potentials issues).
It's also possible to uses any cables supported by openFPGALoader instead of USB-Blaster (required some wires/adapter).

snevindsouza, 2025-03-30:
Hi,

You mentioned that sometimes usb-blaster is not working, I tried it again after a complete system reboot and somehow it worked although it detected a different MAX10 device but still I was able to download my code into the FPGA. However, when I tried it again, I got the same error as before. 

Here's the full WSL log:

```
snevin@WSL:/mnt/c/Users/snevi/OneDrive/Desktop/FPGA/max10text/output_files$ openFPGALoader  -c usb-blaster --probe-firmware /mnt/c/intelFPGA_lite/20.1/quartus/bin64/blaster_6810.hex -v test1.svf
empty
found 1 devices
index 0:
        idcode 0x31820dd
        manufacturer altera
        family MAX 10
        model  10M08SAU169C8G
        irlength 10
FREQUENCY 2.50E+07 HZ
frequency value 2.50E+07 unit HZ25000000
USB-BlasterI has a 24MHz fixed frequency
TRST ABSENT
trst value : ABSENT
ENDDR IDLE
enddr value : IDLE
ENDIR IRPAUSE
endir value : IRPAUSE
STATE IDLE
state value : IDLE

SIR 10 TDI (2CC)
        len   : 10
        tdo   : 0
        tdi   : 12
        mask  : 0
        smask : 0
RUNTEST IDLE 8750003 TCK ENDSTATE IDLE

SIR 10 TDI (203)
        len   : 10
        tdo   : 0
        tdi   : 12
        mask  : 0
        smask : 0
RUNTEST 128 TCK

SDR
        len   : 23
        tdo   : 0
        tdi   : 24
        mask  : 0
        smask : 0

SIR 10 TDI (205)
        len   : 10
        tdo   : 0
        tdi   : 12
        mask  : 0
        smask : 0
RUNTEST 128 TCK

SDR
        len   : 32
        tdo   : 32
        tdi   : 32
        mask  : 32
        smask : 0

SIR 10 TDI (203)
        len   : 10
        tdo   : 0
        tdi   : 12
[... 260 lines cut ...]
        mask  : 0
        smask : 0
RUNTEST 102400 TCK
RUNTEST 512 TCK

SIR 10 TDI (3FF)
        len   : 10
        tdo   : 0
        tdi   : 12
        mask  : 0
        smask : 0
RUNTEST 25000 TCK
STATE IDLE
state value : IDLE
end of SVF file

snevin@WSL:/mnt/c/Users/snevi/OneDrive/Desktop/FPGA/max10text/output_files$ openFPGALoader  -c usb-blaster --probe-firmware /mnt/c/intelFPGA_lite/20.1/quartus/bin64/blaster_6810.hex -v test1.svf
empty
Read error: -7
JTAG init failed with: Unknown device with IDCODE: 0x2f2e2f2e (manufacturer: 0x797 (), part: 0x79 vers: 0x2
```

Could you please let me know what exactly is causing this?

Thank you

trabucayre, 2025-03-30:
Ok. Your cable is a clone based on STM32 instead of FTDI + CPLD. I have never tried this model by myself (I have to get one to check/validates this variant).
Another thing: could you try with a native windows? Sometime WSL do weird things.

snevindsouza, 2025-03-30:
Ok. I see the problem might be the cable. I'll buy the official blaster and try again. Thanks for letting me know 

I had tried it with native Linux and got the same problem but it worked for Cyclone 5. Same with WSL so no difference. However I haven't tried with windows yet. also there's no windows installation guide in **"doc/guide/install.rst"**.  Other sources online led me to MSYS2 or something like that.

Thank you

snevindsouza, 2025-03-30:
HI,

I tried it with native linux again and somehow it worked. Here's the terminal log and I have no clue what's going on but it works for .svf and .pof files.

```
snevin@Linux:/media/snevin/Windows-SSD/Users/snevi/OneDrive/Desktop/FPGA/max10text/output_files$ openFPGALoader -c usb-blaster --detect
empty
index 0:
	idcode 0x31820dd
	manufacturer altera
	family MAX 10
	model  10M08SAU169C8G
	irlength 10
	
snevin@Linux:/media/snevin/Windows-SSD/Users/snevi/OneDrive/Desktop/FPGA/max10text/output_files$ openFPGALoader -c usb-blaster test1.svf
empty
USB-BlasterI has a 24MHz fixed frequency
end of SVF file

snevin@Linux:/media/snevin/Windows-SSD/Users/snevi/OneDrive/Desktop/FPGA/max10text/output_files$ openFPGALoader -v -c usb-blaster test1.svf
empty
found 1 devices
index 0:
	idcode 0x31820dd
	manufacturer altera
	family MAX 10
	model  10M08SAU169C8G
	irlength 10
FREQUENCY 2.50E+07 HZ 
frequency value 2.50E+07 unit HZ25000000
USB-BlasterI has a 24MHz fixed frequency
TRST ABSENT 
trst value : ABSENT
ENDDR IDLE 
enddr value : IDLE
ENDIR IRPAUSE 
endir value : IRPAUSE
STATE IDLE 
state value : IDLE

SIR 10 TDI (2CC) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST IDLE 8750003 TCK ENDSTATE IDLE 

SIR 10 TDI (203) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST 128 TCK 

SDR
	len   : 23
	tdo   : 0
	tdi   : 24
	mask  : 0
	smask : 0

SIR 10 TDI (205) 
	len   : 10
	tdo   : 0
	tdi   : 12
	mask  : 0
	smask : 0
RUNTEST 128 TCK 
[... 331 lines cut ...]
Write Flash: [==================================================] 100.00%
Done
Write UFM0
Write Flash: [==================================================] 100.00%
Done
Write CFM2
Write Flash: [==================================================] 100.00%
Done
Write CFM1
Write Flash: [==================================================] 100.00%
Done
Write CFM0
Write Flash: [==================================================] 100.00%
Done
BSM: 01
DSM Verify: OK
BSM: 01
DSM Verify: OK
BSM: 01
DSM Verify: OK
```

The only thing that is confusing me is that it detects 10M08SAU169C8G but my device is 10M08SAE144C8G so I don't know what's going on here. maybe you do? 

Thank you

trabucayre, 2025-04-16:
I have to check for IDCODE but seems 10M08xxx have all the same value.
Great to see it's for for you