trabucayre/openFPGALoader #539: SRAM programming issue on Tang Primer 25k

tweber-ill, 2025-04-13:
Hi all,

first thanks for your great tool, it works really well with most FPGAs I have! I've only encountered problems with a version of the Tang Primer 25k (PCB version 52300, FPGA: GW5A-LV25 MG121NC1/10, Debugger: BL616C50 with firmware 20241104150): SRAM programming doesn't seem to work at all on it (flash neither, but this is anyway not yet complete as I understand). Everything works fine using the Gowin programmer, though.

Here's the output using the latest version of openFPGALoader from the master branch:
```
~/build/openFPGALoader/build % ./openFPGALoader --detect
empty
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
index 0:
	idcode 0x1281b
	manufacturer Gowin
	family GW5A
	model  GW5A-25
	irlength 8

```

```
~/build/openFPGALoader/build % ./openFPGALoader -b tangprimer25k /tmp/test.fs
empty
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
Parse file Parse /tmp/test.fs: 
Done
DONE
Load SRAM: [==================================================] 100.00%
Done
FAIL
```

```
~/build/openFPGALoader/build % ./openFPGALoader -b tangprimer25k --verbose-level 2 /tmp/test.fs
empty
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
Raw IDCODE:
- 0 -> 0x0001281b
- 1 -> 0xffffffff
Fetched TDI, end-of-chain
found 1 devices
index 0:
	idcode 0x1281b
	manufacturer Gowin
	family GW5A
	model  GW5A-25
	irlength 8
File type : fs
Parse file Parse /tmp/test.fs: 
checksum 0x77bc
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 11072
LoadingRate: 6944444
ProgramDoneBypass: OFF
SPIAddr: 00000000
SecurityBit: ON
idcode: 0001281b
before program sram: displayReadReg 302005fa
	   [ 1] Bad Command
	   [ 3] Timeout
	   [ 4] auto_boot_2nd_fail
	   [ 5] Memory Erase
	   [ 6] Preamble
	   [ 7] System Edit Mode
	   [ 8] Program SPI FLASH directly
	   [10] Non-JTAG configuration is active
	   [21] ECC Error Uncorrectable
	[24:23] CPU Bus Width: no BWD pattern is detected
	[26:25] Retry time sync pattern detect: no retry
	   [28] OTP Reading Done
	   [29] Init Done
Erase SRAM before erase sram: displayReadReg 82fa00e0
	   [ 5] Memory Erase
	   [ 6] Preamble
	   [ 7] System Edit Mode
	   [17] sspi_mode
	   [19] CRC Error
	   [20] ECC Error
	   [21] ECC Error Uncorrectable
	   [22] CMSER IDLE
	[24:23] CPU Bus Width: 8-bit mode
	[26:25] Retry time sync pattern detect: retry one time
	   [31] Auto Erase
pollFlag: fdfffa05 (0)
pollFlag: f4fa47fa (80)
pollFlag: 81602005 (0)
pollFlag: 7d05fa00 (0)
pollFlag: 70026020 (20)
after erase sram: displayReadReg 302005fa
	   [ 1] Bad Command
	   [ 3] Timeout
	   [ 4] auto_boot_2nd_fail
	   [ 5] Memory Erase
	   [ 6] Preamble
	   [ 7] System Edit Mode
	   [ 8] Program SPI FLASH directly
	   [10] Non-JTAG configuration is active
	   [21] ECC Error Uncorrectable
	[24:23] CPU Bus Width: no BWD pattern is detected
	[26:25] Retry time sync pattern detect: no retry
	   [28] OTP Reading Done
	   [29] Init Done
pollFlag: 82fa00e0 (80)
pollFlag: 1005fa05 (0)
pollFlag: 70026020 (0)
pollFlag: 302005fa (80)
pollFlag: 2fa00e0 (20)
after erase sram: displayReadReg 80e00260
	   [ 5] Memory Erase
	   [ 6] Preamble
	   [ 9] auto_boot_1st_fail
	   [21] ECC Error Uncorrectable
	   [22] CMSER IDLE
	[24:23] CPU Bus Width: 8-bit mode
	[26:25] Retry time sync pattern detect: no retry
	   [31] Auto Erase
pollFlag: 7d05fa05 (0)
Load SRAM before write sram: displayReadReg 80e00260
	   [ 5] Memory Erase
	   [ 6] Preamble
	   [ 9] auto_boot_1st_fail
	   [21] ECC Error Uncorrectable
	   [22] CMSER IDLE
	[24:23] CPU Bus Width: 8-bit mode
	[26:25] Retry time sync pattern detect: no retry
	   [31] Auto Erase
Load SRAM: [==================================================] 100.00%
Done
after write sram: displayReadReg 81602005
	   [ 0] CRC Error
	   [ 2] ID Verify Failed
	   [13] Done Final
	   [21] ECC Error Uncorrectable
	   [22] CMSER IDLE
	[24:23] CPU Bus Width: 16-bit mode
	[26:25] Retry time sync pattern detect: no retry
	   [31] Auto Erase
DONE
after program sram: displayReadReg 1005fa00
	   [ 9] auto_boot_1st_fail
	   [11] Bypass
	   [12] i2c_sram_f
	   [13] Done Final
	   [14] Security Final
	   [15] encrypted_format
	   [16] key_right
	   [18] CRC Comparison Done
	[24:23] CPU Bus Width: no BWD pattern is detected
	[26:25] Retry time sync pattern detect: no retry
	   [28] OTP Reading Done
```

The board itself neither shows an error nor success, but just keeps whatever was previously programmed. Lowering the jtag frequency also doesn't help.

Many thanks for any help with that!

trabucayre, 2025-04-16:
Could you try to update the BL616 firmware. some 2024xx firmwares was buggy (see https://trabucayre.github.io/openFPGALoader/guide/troubleshooting.html#cannot-flash-tang-nano-20k-issue-251)
Thanks

trabucayre, 2025-04-16:
Weird. I already updated the firmware for this primer 25k without issues.
Have you others devices connected to your board? Maybe a residual power from another device.

tweber-ill, 2025-04-18:
Yes, just tried with everything else disconnected now, but still to no avail, I still only get the usual two interfaces.

As you've already succeeded updating it, did you do it exactly as they say on their site? Power down, connect 3V3+TDO, connect USB debugger? As I understand there should then be only one single interface visible if it works?

Thanks!

tweber-ill, 2025-04-18:
Update: After changing the pin header, entering the BL616 debug mode finally worked (so it was just a bad connection). And with the new firmware 20250303170 openFPGALoader also works perfectly! Many thanks for the help!

closed 2025-04-25 by tweber-ill

trabucayre, 2025-04-27:
Thanks for the feedback. Great!