trabucayre/openFPGALoader #544: Flashing SPI flash with Gowin  GW5AT-LV60 fails in the first attempt

harbaum, 2025-04-24:
I see this with the current openFPGAloader. I am sure this previously worked a few months ago, so this may be some kind of regression.

What happens is that the first attempt to flash always fails:

```
$ openFPGALoader -f impl/pnr/nanomig_tc60k.fs 
empty
write to flash
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
Parse file Parse impl/pnr/nanomig_tc60k.fs: 
Done
DONE
Jtag frequency : requested 2.50MHz    -> real 2.00MHz   
after program flash: displayReadReg 70026020
           [ 5] Memory Erase
           [13] Done Final
           [14] Security Final
           [17] sspi_mode
        [24:23] CPU Bus Width: no BWD pattern is detected
        [26:25] Retry time sync pattern detect: no retry
           [28] OTP Reading Done
           [29] Init Done
           [30] Wakeup Done
Erase SRAM DONE
Jtag probe limited to %d MHz6000000
Jtag frequency : requested 10.00MHz   -> real 6.00MHz   
JEDEC ID: 0xef4017
Detected: Winbond W25Q64 128 sectors size: 64Mb
JEDEC ID: 0xef4017
Detected: Winbond W25Q64 128 sectors size: 64Mb
RDSR : 0x00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00240000
timeout: 0
write en: Error
Erasing: [==================================================] 100.00%
Fail
Error: write to flash failed
Error: Failed to program FPGA: Error: write to flash failed
```

This leaves the flash in a broken (probably partly erased state) and the FPGA won't successfully boot. But it now can be flashed:

```
$ openFPGALoader -f impl/pnr/nanomig_tc60k.fs 
empty
write to flash
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
Parse file Parse impl/pnr/nanomig_tc60k.fs: 
Done
DONE
Jtag frequency : requested 2.50MHz    -> real 2.00MHz   
after program flash: displayReadReg 16020238
           [ 3] Timeout
           [ 4] auto_boot_2nd_fail
           [ 5] Memory Erase
           [ 9] auto_boot_1st_fail
           [17] sspi_mode
        [24:23] CPU Bus Width: no BWD pattern is detected
        [26:25] Retry time sync pattern detect: no "sync pattern" is found after three times detection
           [28] OTP Reading Done
Erase SRAM DONE
Jtag probe limited to %d MHz6000000
Jtag frequency : requested 10.00MHz   -> real 6.00MHz   
JEDEC ID: 0xef4017
Detected: Winbond W25Q64 128 sectors size: 64Mb
JEDEC ID: 0xef4017
Detected: Winbond W25Q64 128 sectors size: 64Mb
RDSR : 0x00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00240000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
```

If trying to flash other areas of the flash (e.g. at offset 0x400000) this always fails unless a failed attempt to flash the bitstream was made before.

So it seems flashing only works if the FPGA is in unconfigured state. And unsuccessfully trying to flash brings it into this unconfigured state.

trabucayre, 2025-04-27:
Indeed it's a known issue I try to resolve... It's seems only appears when the bitstream is build with secure bit set.
I have to reanalyze jtag stream to see if some special is required

harbaum, 2025-04-27:
I don't think I have any security bit set. 

Also, I cannot flash any binary file like the retro machines ROMs to other offsets in the flash. Therefore I actually don't think this is related to security bits.

I'll try to investigate this a little further.

harbaum, 2025-04-28:
Hmm... I may be wrong that it ever worked. I went back in time and while the symtopms slightly differ between the tags (just hangs vs. exits with error) the problem itself actually persists ...

One issue may theoretically the fact that this is not a real ft2232 but the bl616 emulating it. But I am sure you are aware of that ...

harbaum, 2025-04-28:
Also I don't see any relationship to the security bit being set or not. If I disable the "security bit" tickmark, the resulting bitstream IMHO behaves no different.

trabucayre, 2025-04-28:
Sorry it's a mistake with another Gowin's FPGA.
I will try to update my tang Console with the latest firmware and checking if some modifications related to 138k have not introduces an regression for 60k.

harbaum, 2025-04-28:
Indeed there's a firmware update. I need to try that. Thanks for the hint

trabucayre, 2025-04-29:
With bl616 version 2025041420 no more errors at this level.
But this firmware seems not really stable:
- slow
- jtag communication stall when I open the UART interface (`/dev/ttyUSB1`) during flash programming
- same error as mentioned previously when the UART interface is used before flash programming

harbaum, 2025-04-29:
Interesting. Over at the Tang console discord was a report that the original Gowin programmer is not able to write the flash with this latest firmware, anymore. After your report I am tempted to give it a try, anyways as I always use openFPGAloader.

harbaum, 2025-04-30:
With the updated firmware it also works for me albeit super slow. This IMHO means that this is an issue with the bl616 and not openfpgaloader. So this can IMHO be closed.

trabucayre, 2025-04-30:
Thanks for your feedback. I'm agree: problem seems related to the USB interface