trabucayre/openFPGALoader #550: Windows 'openFPGALoader --scan-usb' fails with some devices

zapta, 2025-05-19:
OpenFPGALoad --scan-usb fails consistently on windows with some off the shelf FPGA boards and works with others (details below). On mac osx, openFPGALoader works well with all of these boards.  

```
# -- Windows version
[OSS CAD Suite] C:\Users\user\Downloads\oss-cad-suite>ver
Microsoft Windows [Version 10.0.19045.5854]

# -- openFPGALoader's version. 
# -- The binary is from yosys nightly release
# -- https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2025-05-19
#
[OSS CAD Suite] C:\Users\user\Downloads\oss-cad-suite>openFPGALoader -V
openFPGALoader v0.13.1

# -- Works properly with Upduino 
[OSS CAD Suite] C:\Users\user\Downloads\oss-cad-suite>openFPGALoader --scan-usb
empty
Bus device vid:pid       probe_type manufacturer  serial   product
002 008    0x0403:0x6010 FTDI2232   tinyVision.ai FT94RQ8V UPduino v3.1c

# -- Fails with Alhambra ii
[OSS CAD Suite] C:\Users\user\Downloads\oss-cad-suite>openFPGALoader --scan-usb
empty
Error: can't open device with vid:vid = 0x0403:0x6010. Error code -5 Entity not found
Bus device vid:pid       probe_type manufacturer serial product

# -- Fails with Tang Nano 9K
[OSS CAD Suite] C:\Users\user\Downloads\oss-cad-suite>openFPGALoader --scan-usb
empty
Error: can't open device with vid:vid = 0x0403:0x6010. Error code -5 Entity not found
Bus device vid:pid       probe_type manufacturer serial product
```

zapta, 2025-05-19:
And here are the same three boards on mac osx:

```
~/Downloads/oss-cad-suite/bin$ ./openFPGALoader -V
openFPGALoader v0.13.1

~/Downloads/oss-cad-suite/bin$ ./openFPGALoader --scan-usb
empty
Bus device vid:pid       probe type      manufacturer serial               product
000 005    0x0403:0x6010 FTDI2232        tinyVision.ai FT94RQ8V             UPduino v3.1c
000 004    0x0403:0x6010 FTDI2232        SIPEED       FactoryAIOT Pro      JTAG Debugger
000 003    0x0403:0x6010 FTDI2232        AlhambraBits none                 Alhambra II v1.0A - B09-335
```

trabucayre, 2025-05-19:
I have to check. It's with WSL?
Windows is not really to me.
According to the list the only board I have is the sipeed.

zapta, 2025-05-19:
1. That's a regular Winodws, no WSL, with openFPGALoader.exe from the yosys project.

2. This is the Tang board mentioned in my original post https://wiki.sipeed.com/hardware/en/tang/Tang-Nano-9K/Nano-9K.html

3. We program the alhambra-ii with openFPGALoader as a generic ice40 and it works on mac osx.

4. I think that openFPGALoader should not crash on --scan-usb, even if a FPGA board that it doesn't support is connected to the host.

5. Any idea or a pointer on how to build openFPGALoader for windows?

zapta, 2025-05-20:
@trabucayre since you mentioned that windows is not your system of choice, the daily builds of oss-cad-suite includes the latest code from this open openFPGALoader repo so instead of building for windows you can just pick the windows package from the latest oss-cad-suite build https://github.com/YosysHQ/oss-cad-suite-build/releases

After you unpack the package, you will find a batch file called 'environment' which will setup your cmd shell path to have openFPGALoader in the path.

trabucayre, 2025-05-20:
I have installed oss-cad-suite and tested with a powershell: I'm able to enumerates the tang Nano 9k. The first thing I have done was to uses zadig to switch both interface to WinUSB driver.
But i don't know why but with WSL nor openFPGALoader nor lsusb are able to detect the board

zapta, 2025-05-20:
Thanks @trabucayre, setting the Zadig to WinUSB fixed the problem with the two boards (I thought it was already configured), and without this setting openFPGALoader just skips that devices and doesn't abort which is also good. All is good.

Generally we recommend to set only interface 0 because interface 1 may be used by the app for communicating with the FPGA board.

https://github.com/FPGAwars/apio/blob/e2bb0f46f03dddbe580e7f0a250b55746897500e/apio/managers/drivers.py#L72

Thanks for your quick help, closing this one.

closed 2025-05-20 by zapta