trabucayre/openFPGALoader #559: Consistent 'Done' for 'Wait for CDONE'

zapta, 2025-06-04:
Please make the `DONE` of `Wait for CDONE` consistent with the `Done` of the previous steps. This will also provide a final 'Done' at the end  of the run.

Before the fix:
```
openFPGALoader --verify -b ice40_generic --vid 0403 --pid 6010 --busdev-num 0:3 _build/default/hardware.bin
...
start addr: 00000000, end_addr: 00030000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
Wait for CDONE DONE
```

After the fix:
```
openFPGALoader --verify -b ice40_generic --vid 0403 --pid 6010 --busdev-num 0:3 _build/default/hardware.bin
...
start addr: 00000000, end_addr: 00030000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Done
Wait for CDONE
Done
```

trabucayre, 2025-06-08:
Done with https://github.com/trabucayre/openFPGALoader/commit/347088efb4ed9c6bacdab972b0837d076c184efb

zapta, 2025-06-12:
Perfect. Thanks!

```
openFPGALoader --verify -b ice40_generic --vid 0403 --pid 6010 --busdev-num 0:6 _build/default/hardware.bin
empty
Cable VID overridden
Cable PID overridden
Jtag frequency : requested 6.00MHz    -> real 6.00MHz   
Parse file DONE
JEDEC ID: 0xef4016
Detected: Winbond W25Q32 64 sectors size: 32Mb
00000000 00000000 00000000 00
start addr: 00000000, end_addr: 00030000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Reading: [==================================================] 100.00%
Done
Reset and Wait for CDONE:
Done
```

closed 2025-06-13 by trabucayre