trabucayre/openFPGALoader #566: Unable to write to microchip SST26VF064B

moffel, 2025-06-28:
```
openFPGALoader -c gatemate_pgm --verify -f .\bitfile.bit
empty
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Detected: microchip SST26VF064B 128 sectors size: 64Mb
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Verifying write (May take time)
Read flash : [==================================================] 100.00%
Fail
Verification failed at 0
```

Reading back the flash reveals that it is completely empty.

This might be related to the fact that this chip requires a global unlock after power up before any write. The global unlock is implemented for the 32Mb version of this flash at spiFlash.cpp:360 `if (_jedec_id == 0xbf2642bf) {  // microchip SST26VF032B` but seems to be missing for the 64Mb version.

The requirement to isse a global unlock before any write should probably be a flash property in spiFlashdb.hpp.

In addition it would be nice to add a flag or resuse --unprotect-flash to also issue the global protect unlock command. This command is actually not microchip specific and can be found on winbound chips as well. (W25Q128JV)

It would be great if we could have some very low level spi command issuing flags to diagnose problems like this one.

trabucayre, 2025-06-30:
I have pushed a first commit to handled global_lock via spiFlashdb entries. for SST26 this unlock is always mandatory.
Remaining part is to check the bit for all Flash chip with this feature to decide if it's required to unlock in parallel to the classic approach.

trabucayre, 2025-07-01:
Have you some link to documentation on how to build an application natively? I have already built openFPGALoader with MSYS or WSL. I'm interested to see errors because openFPGALoader build successfully on Linux, MacOS, MSYS and WSL.