trabucayre/openFPGALoader #572: openFPGALoader detect command hangs with GW2AR-18C and FTDI FT2232H

I0Y0I, 2025-07-14:
I'm trying to program a Gowin GW2AR-18C FPGA using openFPGALoader and an FTDI FT2232H-based USB cable. When I run openFPGALoader --detect, the command hangs after displaying the following output:

```bash
❯ openFPGALoader --detect
empty
No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz    -> real 6.00MHz
```

The program then freezes, and there's no further output or response.

I've already verified the connection to the GW2AR-18C using Gowin Programmer, which successfully detects and connects to the board, confirming the hardware setup and cable functionality.

I've also set up the udev rules according to the openFPGALoader documentation to ensure proper USB device permissions. My dmesg output when connecting the cable shows:

```
[ 367.046008] usb 1-9: new high-speed USB device number 9 using xhci_hcd
[ 367.170348] usb 1-9: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 5.00
[ 367.170362] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 367.170368] usb 1-9: Product: Dual RS232-HS
[ 367.170372] usb 1-9: Manufacturer: FTDI
[ 367.170375] usb 1-9: SerialNumber: 2023081413
```

This confirms the `idVendor=0403` and `idProduct=6010` (FTDI FT2232H) is being recognized by the system.

Environment details:

    Operating System: Linux 6.15.6-2-cachyos (64-bit)

    openFPGALoader Version: openFPGALoader v0.13.1

    FPGA Board: Gowin GW2AR-18C

    JTAG Debugger/Cable: FTDI FT2232H-based USB cable (identified as idVendor=0403, idProduct=6010 and Product: Dual RS232-HS)

Given that Gowin Programmer works correctly with the same setup, it suggests the issue might be specific to openFPGALoader's interaction with this particular FTDI device or the GW2AR-18C's JTAG interface.

Troubleshooting steps I've already taken:

    Verified connection with Gowin Programmer (successful).

    Configured udev rules for permissions.

    Confirmed FTDI device recognition via dmesg.

Any insights or suggestions on how to debug this further would be greatly appreciated!

I0Y0I, 2025-07-14:
I've run openFPGALoader --detect under gdb and have captured the backtrace when the program hangs and I interrupt it with Ctrl+C.

The output from gdb is as follows:
```
(gdb) r --detect
Starting program: /usr/bin/openFPGALoader --detect
Function(s) ^std::(move|forward|as_const|(__)?addressof) will be skipped when stepping.
Function(s) ^std::(shared|unique)_ptr<.*>::(get|operator) will be skipped when stepping.
Function(s) ^std::(basic_string|vector|array|deque|(forward_)?list|(unordered_|flat_)?(multi)?(map|set)|span)<.*>::(c?r?(begin|end)|front|back|data|size|empty) will be skipped when stepping.
Function(s) ^std::(basic_string|vector|array|deque|span)<.*>::operator.] will be skipped when stepping.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
empty
No cable or board specified: using direct ft2232 interface
[New Thread 0x7ffff75ff6c0 (LWP 3635)]
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
^C
Thread 1 "openFPGALoader" received signal SIGINT, Interrupt.
0x00007ffff76af6b2 in __syscall_cancel_arch () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff76af6b2 in __syscall_cancel_arch () from /usr/lib/libc.so.6
#1  0x00007ffff76a34a8 in __internal_syscall_cancel (a1=<optimized out>, a2=<optimized out>, a3=<optimized out>, a4=0, a5=0, 
    a6=0, nr=7) at cancellation.c:49
#2  __syscall_cancel (a1=<optimized out>, a2=<optimized out>, a3=<optimized out>, a4=a4@entry=0, a5=a5@entry=0, 
    a6=a6@entry=0, nr=7) at cancellation.c:75
#3  0x00007ffff772b1ba in __GI___poll (fds=<optimized out>, nfds=<optimized out>, timeout=<optimized out>)
    at ../sysdeps/unix/sysv/linux/poll.c:29
#4  0x00007ffff7f75399 in poll () at /usr/include/bits/poll2.h:44
#5  usbi_wait_for_events () at os/events_posix.c:266
#6  handle_events () at /usr/src/debug/libusb/libusb/libusb/io.c:2254
#7  0x00007ffff7f762a7 in libusb_handle_events_timeout_completed () at /usr/src/debug/libusb/libusb/libusb/io.c:2371
#8  0x00007ffff7f764af in libusb_handle_events_completed () at /usr/src/debug/libusb/libusb/libusb/io.c:2469
#9  0x00007ffff7f765c4 in sync_transfer_wait_for_completion () at /usr/src/debug/libusb/libusb/libusb/sync.c:55
#10 do_sync_bulk_transfer () at /usr/src/debug/libusb/libusb/libusb/sync.c:200
#11 0x00007ffff7f78703 in libusb_bulk_transfer () at /usr/src/debug/libusb/libusb/libusb/sync.c:286
#12 0x00007ffff7f5c555 in ftdi_read_data () at /usr/src/debug/libftdi/libftdi1-1.5/src/ftdi.c:2048
#13 0x0000555555832e58 in FTDIpp_MPSSE::mpsse_read ()
    at /usr/src/debug/openfpgaloader/openFPGALoader-0.13.1/src/ftdipp_mpsse.cpp:552
#14 0x0000555555829687 in FtdiJtagMPSSE::writeTDI ()
    at /usr/src/debug/openfpgaloader/openFPGALoader-0.13.1/src/ftdiJtagMPSSE.cpp:273
#15 0x000055555589d94e in Jtag::read_write () at /usr/src/debug/openfpgaloader/openFPGALoader-0.13.1/src/jtag.cpp:364
#16 Jtag::detectChain(unsigned int) [clone .isra.0] () at /usr/src/debug/openfpgaloader/openFPGALoader-0.13.1/src/jtag.cpp:196
#17 0x0000555555825f3b in Jtag::Jtag () at /usr/src/debug/openfpgaloader/openFPGALoader-0.13.1/src/jtag.cpp:169
#18 0x00005555556eda37 in main () at /usr/src/debug/openfpgaloader/openFPGALoader-0.13.1/src/main.cpp:462
(gdb) 
```

trabucayre, 2025-07-21:
I have tested (for #573) real ftdi usage (yes it's an FT232H and not a FT2232H) without issues.
According to your dump issue is between libusb and libftdi when a read is performed: the FTDI doesn't answer or the event is not correctly catched/used by the library.
Is it possible to you to test with another distribution like debian or ubuntu: the idea is to see if the issue is related to the cable or to the distribution?

I0Y0I, 2025-07-24:
I tried the same operation on Windows today and got the same result, which leads me to believe it's a cable issue. The debugger I'm using is an FTDI interface emulated by a CH32V305, so it's possible some functionalities aren't fully implemented.

To debug this, I used Wireshark to capture and compare packets from the gowin-eda-programmer, and the results are below. 

633 to 672 are caused by command:
```
openFPGALoader --verbose-level 2 --detect
```

28703 to 28860 are caused by command:
```
/opt/gowin-eda-programmer/bin/programmer_cli --scan
```
and it returns
```
 Scanning!
 Target Cable: Gowin USB Cable(FT2CH)/0/None/null@2.5MHz
Device Info:
 	Family: GW2ANR  
	Name: GW2A-18C GW2AR-18C GW2ANR-18C  (One of them)  
	ID: 0x0000081B  
 1 device(s) found!
 Cost 0.55 second(s)
```

Since I'm not familiar with FTDI, I suspect the problem might be related to event 28857 (`Bit Mode: switch off bitbang mode, back to regular serial / FIFO (0x00)`).

Please let me know if my assumption is correct. I would be very grateful for any suggested solutions or recommendations for open-source test cables.

[wireshark_results.tar.gz](https://github.com/user-attachments/files/21413476/wireshark_results.tar.gz)

I0Y0I, 2025-08-01:
Today, I purchased the RV debugger from the debugger list and everything is now working correctly. This indicates that the issue was a debugger bug, not a program bug. Thanks for the support!

trabucayre, 2025-08-02:
FTDI clone are sometime a real nightmarre because protocol may not be fully implemented, FIFO size too small or implementation too slow.
Usually when communication goes wrong it's always better to test with known to work probe. One good device/protocol is the Pico DirtyJtag. device is low cost and it has a dedicated protocol instead of an MPSSE implementation.