trabucayre/openFPGALoader #578: esp32s3 cable fails to open

mohammed-nurulhoque, 2025-08-12:
Trying to use an esp32s3 with esp-usb-bridge firmware to program a colorlight 5a-75b fpga. 
```
empty
fails to open esp_usb_jtag device vid:pid 0x303a:0x
JTAG init failed with: std::exception

```

I compiled openFPGALoader from main. Currently in this repo, it puts the PID of esp32s3 as 1001 but mine shows 1002 so I patched that.and still got the error above.

I can see the device in lsub as `Bus 001 Device 125: ID 303a:1002 Espressif Systems Co. Ltd. ESP USB Bridge`
And openocd is able to see the fpga through the bridge
```
$ openocd -f setup.cfg -c "init; svf blink.svf; exit"
Open On-Chip Debugger 0.12.0+dev-02123-g306114954 (2025-08-12-16:12)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: Device found. Base speed 4800KHz, div range 1 to 1
Info : clock speed 4800 kHz
Info : JTAG tap: ecp5.tap tap/device found: 0x41111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x4)
```
but is failing for me for other reasons https://github.com/YosysHQ/prjtrellis/issues/252

Anyway, that shows the device is definitely connected.

mohammed-nurulhoque, 2025-08-13:
Oh I used https://github.com/espressif/esp-usb-bridge

emard, 2025-08-25:
> I have pushed this commit
> esp32-s3 with espressif firmware is correctly seen/uses but the FPGA is
> not seen (it's also true with openocd).
> My pinout is based on @emard <https://github.com/emard> firmware -> maybe
> something wrong with the firmware's configuration.
> Could you try in your side?
> Thanks

can yo send or upload o net firmware binary so i can download it and try
with my hardware

emard, 2025-08-25:
tricky things there is how to configure pins so that internal jtag is
detached from esp32 CPU JTAG
and routed outside to hardware pins of JTAG to FPGA
This is sofware routing of JTAG port and would be very interesting
if made to work

I know your mentioned source in C but I have never tried it.
I tried to make micropython version of that instead but currently
micropython is not yet ready, I issued and waiting for support.

Using my arduino source I succeded to hardware-route
JTAG of S3 as you know. I tried the same source on C3
and it doesn't work, it compiles ok but connects to internal
esp32 CPU JTAG instead of exposing external JTAG
although I have set exposing C3 registers but something
is missing

trabucayre, 2025-08-27:
Small update: with https://github.com/espressif/esp-usb-bridge tag v1.1.0 everything is working with openFPGALoader master branch.
Look like some refactoring have broken the code.

emard, 2025-09-01:
Great!

I'll try to compile the code maybe even locate a patch that breaks it

emard, 2025-09-24:
> @emard <https://github.com/emard> I'm not sure it's relevant to search
> the commit introducing this regression. People working on this repository
> seems to prepare a rework: maybe better to wait for next release?

I agree, let them release new then we'll see how it works
they are probably already aware for things to fix