trabucayre/openFPGALoader #585: Lattice CertusPro-NX external flash Read ID fail

MSOcu, 2025-09-08:
Hello,
let me first thank you for your work and your awsome tool!

I've found the following issue:
I'm using a Lattice CertusPro-NX with external SPI flash. My issue is, that I can flash the FPGA 1-2 times back to back and it is working.
After the succesful tries, the FPGA seems to get stuck in the Boot mode and wont start. I can then still access the FPGA SRAM but not the flash.
It seems like the Flash ID could not be read anymore. 
A reset via the openFPGAloader is also not possible or seems not to work. I have to manually toggle the PROGRAMN pin or pull it down, until openFPGAloader is startet again. Then the flash procedure works as intendet.

I've also testet to detect the external flash via openFPGALoader -c ft232 --ftdi-channel 0 --detect --external-flash . this command also puts the FPGA in the faulty state.

Attached the verbose output in case of an error:
openFPGALoader -c ft232 --ftdi-channel 0 -f Baseboard_Bringup_impl_1.bit -v
empty
write to flash
Can't read iSerialNumber field from FTDI: considered as empty string
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
found 1 devices
index 0:
        idcode 0x10f4043
        manufacturer lattice
        family CertusProNX
        model  LFCPNX-100
        irlength 8
File type : bit
IDCode : 10f4043
displayReadReg
        Config Target Selection : 0
        Done Flag
        BSE Error Code
                No err
        BSE Error 1 Code for previous bitstream execution
                No err
        SFDP Timeout
        INITN
        Authentication mode:
                No Auth
Open file DONE
Parse file b3beffff
DONE
bitstream header infos
Architecture: jd5d00
Bits: 15005224
Bitstream CRC: 0xD14D
Bitstream Status: Final Version 4.6
Cols: 892
Date: Mon Sep  8 08:30:55 2025
Design name: Baseboard_Bringup_impl_1.udb
Part: LFCPNX-100-12LFG672
Readback: Off
Rows: 16822
Security: On
Version: Radiant Software (64-bit) 2025.1.0.39.0
idcode: 010f4043
Enable configuration: DONE
SRAM erase: DONE
0 0 0 0 Read ID failed
Refresh: DONE
Error: Failed to program FPGA: std::exception

Case when working:

openFPGALoader -c ft232 --ftdi-channel 0 -f Baseboard_Bringup_impl_1.bit                                                                                                                                          
empty
write to flash
Can't read iSerialNumber field from FTDI: considered as empty string
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Open file DONE
Parse file b3beffff
DONE
Enable configuration: DONE
SRAM erase: DONE
JEDEC ID: 0xc2201a
Detected: Macronix MX25L51245G 1024 sectors size: 512Mb
00000000 00000000 00000000 40
start addr: 00000000, end_addr: 001e0000
Erasing: [==================================================] 100.00%
Done
Writing: [==================================================] 100.00%
Done
Refresh: DONE

MSOcu, 2025-10-23:
Thank you for your respose, also sorry for the late follow up.
It was on a custom board. Today I tried it on an eval board with the same issue. I have to manually put the FPGA into reset state, before I could flash it.
Also the reset after flash is not working.

trabucayre, 2025-12-30:
Hi and sorry for delay. I have received an EVN board and was able to fix this issue.
With the patch serie:
- https://github.com/trabucayre/openFPGALoader/commit/dd7b74d2f5842d91d8e8f8e44757c52410fe6d07
- https://github.com/trabucayre/openFPGALoader/commit/6f920360fcc6d990bb84b9c87a799b0a80dc424e
- https://github.com/trabucayre/openFPGALoader/commit/6f920360fcc6d990bb84b9c87a799b0a80dc424e

Both mode are working without having to power cycle the board or to deal manually with signals.

trabucayre, 2026-01-02:
@MSOcu Is it good to you?