trabucayre/openFPGALoader #589: Error when loading Virtex5

mert-ekren, 2025-09-25:
Greetings I was using Virtex5 to load MCS file for bootloader of a virtex5 series fpga and i get this error JTAG init failed with: Unknown device with IDCODE: 0x32ece093 (manufacturer (Xilinx), part: 0x17 vers: 0x3, i know that it doesnt support virtex 5 is there an extension posibility, i am mainly using the trimode ethernet for ee5410.

trabucayre, 2025-09-25:
openFPGALoader needs to now some basics informations about the device. This is done via part.hpp where for all models the IDCODE (here: 0x32ece093) is associated with a manufacturer, a family, a model and an IR length.
Since no virtex5 are present in openFPGALoader, the IDCODE is unknown and consequently openFPGALoader has no informations.

To adds support you have first to update part.hpp to add a new entry for your device and test as it. It also maybe possible some adjustements are required since xilinx.cpp has no informations about virtex5.
First of all is to add IDCODE in part.hpp and test if basic functionnalities are working (loading into SRAM).

mert-ekren, 2025-09-28:
Family = Virtex5
Device = xc5vsx95t
Package = ff1136
Speed = -2